RehabMan/OS-X-ACPI-Battery-Driver #8: after update to 10.10.2 - i am getting powerd message in the log every 30 seconds.

tluck, 2015-01-29:
here is cut and paste from /var/log/system.log - is this a DSDT issue?

Jan 29 11:33:00 Toms-Mac.local powerd[29]: Failed to read current rating(0xe00002f0)
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::pollingTimeOut called
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::pollBatteryState: path = 1
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBatteryManager::getBatterySTA called
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatterySTA: battery_status = 0x1f
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBatteryManager::getBatteryBIF called
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBatteryManager::getBatteryBIF: validateObject return 0x0
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBIF: acpibat_bif size = 13
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBIF: fPowerUnit       = 0x0
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBIF: fDesignCapacity  = 84240
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBIF: fMaxCapacity     = 77720
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBIF: fBatteryTech     = 0x1
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBIF: fDesignVoltage   = 10800
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBIF: fCapacityWarning = 3886
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBIF: fLowWarning      = 200
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBIF: fDeviceName      = 'Battery'
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBIF: fSerialNumber    = 'xxxxx'
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBIF: fType            = 'LION'
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBIF: fManufacturer    = 'LENOVO'
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBatteryManager::getBatteryBST called
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBST: acpibat_bst size = 4
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBST: fPowerUnit       = 0x0
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBST: currentStatus    = 0x0
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBST: fCurrentRate     = 0
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBST: fCurrentCapacity = 76620
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBST: fCurrentVoltage  = 12068
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBST: Calculating for WATTS
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBST: fCurrentRate = 0
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBST: fCurrentCapacity = 7094
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::setBatteryBST: fAverageRate = 0
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery: Battery is charged.
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::constructAppleSerialNumber called
Jan 29 11:33:30 Toms-Mac kernel[0]: AppleSmartBattery::rebuildLegacyIOBatteryInfo called
Jan 29 11:33:30 Toms-Mac.local powerd[29]: Failed to read current rating(0xe00002f0)

RehabMan, 2015-02-04:
It seems this is a FakeSMC (key) issue.

tluck, 2015-02-05:
Any thoughts on what key to add?

tluck, 2015-02-09:
This was fixed by adding the SMC key ACID into FakeSMC.

closed 2015-02-09 by tluck

Hs-Yeah, 2015-02-21:
@tluck Can you tell me how to add the SMC key AID into FakeSMC ?

tluck, 2015-02-21:
i added this key to the Keys section of FakeSMC.kext/Contents/Info.plist
                                        <key>ACID</key>
                                        <array>
                                                <string>ch8*</string>
                                                <data>
                                                ut7DNVAFEGM=
                                                </data>
                                        </array>

Hs-Yeah, 2015-02-23:
@tluck Thank you! I added the Key and rebuild the kextcache but it didn't work. :(

RehabMan, 2015-02-25:
It works -- @Hs-Yeah... you must have added it incorrectly.

Eventually, some version of this will appear in my FakeSMC fork.

Hs-Yeah, 2015-02-25:
@RehabMan @tluck I added it here. Did I make it wrong?
[image]
My SMBIOS is set to MacBookPro9,2 and my AC Adapter is 90W.

tluck, 2015-02-25:
that's seems right to me. if you look in About this Mac/System Report/Hardware/Power
does your AC charger Information look like this?

AC Charger Information:

  Connected:	Yes
  ID:	0x0100
  Wattage (W):	85
  Revision:	0x0000
  Family:	0x00ba
  Serial Number:	0x0035c3de
  Charging:	No

Hs-Yeah, 2015-02-25:
@tluck Nope, only

Connected: Yes
Charging: No

I think the ID things will not appear until I make the powerd message gone.

tluck, 2015-02-25:
well then you really don't seem to have the ACID key there? 
assuming your FakeSMC is in /S/L/E,  you can force a kextcache rebuild by:
```    sudo touch /System/Library/Extensions
````
and then restart.

or to read your keys run:
```smcutil -l | more
````

Hs-Yeah, 2015-02-25:
@tluck @RehabMan 
I reinstall the FakeSMC.kext with my edit of Info.plist in it using the Kext Wizard and then using the command @tluck gave me to rebuild the kextcache.
After reboot the problem was fixed.
Thank you!