RehabMan/OS-X-ACPI-Battery-Driver #9: Battery on 10.10.3

walkman8196, 2015-04-15:
Hello RehabMan.
I use your kext and dsdt patch (battery_HP-DV7-6xxx.txt) for the battery
Before version 10.10.3 everything worked perfectly.
And this problem still occurs in some people.

You can see what is the reason?
I can provide reports on demand.

bySabi, 2015-04-23:
Same problem here with 10.10.3.

RehabMan, 2015-04-26:
Post ioreg and logs from system.log (with debug version).

walkman8196, 2015-04-26:
https://cloud.mail.ru/public/7nhSJL994Swj/system.log
https://cloud.mail.ru/public/5fwYoSUXrJ55/MacBook%20Pro%C2%A0%E2%80%94%20%D0%9A%D0%BE%D0%BD%D1%81%D1%82%D0%B0%D0%BD%D1%82%D0%B8%D0%BD.ioreg

RehabMan, 2015-04-27:
Your DSDT is reporting that battery is not present:
[code]
Apr 26 09:06:52 localhost kernel[0]: AppleSmartBattery::setBatterySTA: battery_status = 0xf
Apr 26 09:06:52 localhost kernel[0]: AppleSmartBattery: !fBatteryPresent
[/code]

_STA should report 0x1F when battery is installed.  Refer to ACPI spec.

walkman8196, 2015-04-27:
new ioreg from 2.1 and a new log
I ran the laptop on battery without AC adapter but in the log still battery_status = 0xf
https://cloud.mail.ru/public/4QVmKmtGi7PA/data.zip

RehabMan, 2015-04-27:
You have DSDT issues.  You probably need patches.  Read/follow my guide: http://www.tonymacx86.com/yosemite-laptop-support/116102-guide-how-patch-dsdt-working-battery-status.html

RehabMan, 2015-04-27:
Note also that ACPIDebug.kext can be useful for debugging DSDT problems: https://github.com/RehabMan/OS-X-ACPI-Debug

walkman8196, 2015-04-28:
Rehabman Hello, thanks for the advice.
did not look for the cause, took a clean patch dsdt Hpet and battery, start, and it worked.
Thanks for the advice,
and for your time

closed 2015-04-28 by walkman8196