RehabMan/OS-X-ACPI-Battery-Driver #11: Changed from Chameleon to Clover, no more battery icon

carlosfvp, 2015-07-03:
Hi, I experienced a little bug but I wasn't sure where to post the issue, here or open a ticket on Clover SourceForge.

As I have seen you more active and I saw all your configs are made with Clover as boot maybe you experienced this issue.

I have originally generated a DSDT with MaciASL and applied only patches from your repo. The battery icon was working really well with Chameleon but since I have changed to Clover the battery icon disappeared.

HWMonitor show the battery status wich is weird.

Your Battery Kext loaded correctly and the DSDT is also loading fine because I also applied the fix for the screen brightness and the brightness bar appear on Settings>Displays.

Also the FakeSMC (I know this is a different driver) is also not working anymore. I noticed this because the Voltage, CPU usage doesn't  show up on HWMonitor.

I am attaching some screenshots from IOReg wich shows that everything should be ok but it's not.

[image]
[image]
[image]

carlosfvp, 2015-07-03:
I've been looking some logs and Clover still loads NullCPUPowerManagement.kext also a FakeSMC that doesn't exist beacuse I have deleted both kext from /S/L/E folder. Maybe this isn't loading the Battery kext too. I am sorry, this is not the right place to place this issue but have you experienced this bug with Clover? Thank you.

carlosfvp, 2015-07-04:
Searching for the error of duplicate FakeSMC wich I also have, the solution was reinstall OSX Yosemite package (bundle from Hackintosh Zone) again without NullCPUPowerManagement and FakeSMC marked. This behavior is really weird beacuse I had already deleted older FakeSMC and NullCPUPowerManagement but they were still being loaded when I changed to Clover! Anyway, reinstalling without those kexts solver my issue at all. At least the OSX installer let you install the system and keep your personal files intact. Sorry for posting this here but I had no where else to ask for help. Let's leave this here for future reference.

Any idea why NullCPU and FakeSMC were still loading even when they were deleted? I have rebuilt the cache and even running Clover with NoCache option marked.

You can close this issue after any last comment you might want to add. Sorry for the long and maybe unnecessary issue opened here.

RehabMan, 2015-07-30:
No idea without seeing your Clover config, EFI/Clover folder.

At any rate, not related to ACPIBatteryManager.kext.

closed 2015-07-30 by RehabMan