RehabMan/OS-X-ACPI-Battery-Driver #12: Battery at 0%, all the time.

lolman8776, 2016-08-16:
Hi. Running OS X Yosemite on ASUS Vivobook s500ca here, and trying to use this kext for battery. However, although your kext makes the battery indicator appear, it is always at 0%. Although this is my first attempt at hackintosh, I suspect it has something to do with DSDT needing to be patched. I don't know what it is (and I don't think I have one), but I think I saw something about that in Chameleon Wizard (I'm using Chameleon as a bootloader). If my DSDT does need to be patched, can you please walk me through it?

the-braveknight, 2016-08-18:
http://www.tonymacx86.com/threads/guide-how-to-patch-dsdt-for-working-battery-status.116102/

lolman8776, 2016-08-19:
[This](https://github.com/RehabMan/Laptop-DSDT-Patch/blob/master/battery/battery_ASUS-N55SL.txt) patch works for me, but it makes my Hackintosh go from the speed of a normal mac to incredibly slow, and get graphical errors like the mouse doubling, top bars (the ones that have the close buttons) not appearing, etc. A full reinstall is required to get it back up to its normal speed, but that removes the DSDT patch (duh..). How can I stop it from becoming so slow?

lolman8776, 2016-08-19:
I have come to the conclusion that Chameleon Wizard is causing the slowness, as modifying anything using it will make the Hackintosh slow. This means even Graphics Mode. I'll reinstall and try adding the DSDT to my boot.org.chameleon.plist (I hope I got that right) manually. If it works, I'll close this issue. Otherwise, I'll make a new issue on the relevant repository.