RehabMan/OS-X-ACPI-Battery-Driver #15: Discharging View with 2 Batteries on 10.11.6 stays red on battery switch

herrSei, 2016-12-04:
I'm using the Thinkpad T440s with 2 batteries without the battery-merge hack. its running quite well. except a minor bug: when one battery is empty, its shown with a red battery indicator symbol in the menu bar as desired. but after bios switch to the second (fully charged) battery, it keeps red instead of switching to a black (non critical) feedback symbol, although it show the percentage correctly.

tluck, 2016-12-05:
well i wonder why you dont use the dual battery fix? this patch worked great for me.
[4_battery_fan.txt](https://github.com/RehabMan/OS-X-ACPI-Battery-Driver/files/631305/4_battery_fan.txt)

herrSei, 2016-12-05:
Hey tluck. I tried but i didn't worked out in first place. However, this patch worked like a charm. Thank you very much!

closed 2016-12-05 by herrSei