RehabMan/OS-X-ACPI-Battery-Driver #17: 10.12.4 - crashes Activity Monitor

wmchris, 2017-03-28:
See: http://www.insanelymac.com/forum/topic/322196-macos-sierra-10124-is-out/?p=2393652

RehabMan, 2017-05-11:
It is not a problem in ACPIBatteryManager.kext itself.
Problem is somewhere else in the system...

wmchris, 2017-05-11:
many many many people have this problem on multiple different notebooks and it's fixed after removal of the ACPIBatteryManager.

RehabMan, 2017-05-11:
I realize that.
But it is not a bug in ACPIBatteryManager.kext.
It will happen with any non-Apple battery status kext.

RehabMan, 2017-08-27:
This is fixed in version 1.71.
The fix was to change back to AppleSmartBatteryManager and AppleSmartBattery for the class names, as Apple seems to have some dependencies on that.

The class names were changed to conform to Apple guidelines in commit 0345093 back in 2013.

Thanks to oc_bot (on tmx) for reporting that version 1.32 was fixing this issue.  I was able to determine that also 1.33 was working, and that the change happened between 1.33 and 1.40.

closed 2017-08-27 by RehabMan