RehabMan/OS-X-ACPI-Battery-Driver #20: Strange symptom on laptop with AMT

xtreme1690, 2017-08-09:
System: HP Elitebook 2570p - Intel Core i5 3360m

It seems anytime AMT is involved in the BIOS, i cant get this to work. It will work and the battery % is correct and when AC is plugged in it will work correctly when i tell the bios to "Unconfigure AMT on next reboot" but as soon as a next reboot/shutdown takes place this goes back to not working!

What happens? Well it seems AMT is causing the battery % and status to be reported incorrectly. When plugged in It will also say "BATTERY IS NOT CHARGING"

There doesnt seem to be a way to permanently disable AMT either despite HP's own document stating howto get into the firmware management (CTRL+P)

Have no problems on a same model HP 2570p unit with a I5 processor (without 'V-Pro')

RehabMan, 2017-08-27:
Please provide "Problem Reporting" files in the ProBook guide thread.

RehabMan, 2017-08-29:
Closing due to no response, no followup.

closed 2017-08-29 by RehabMan