RehabMan/OS-X-ACPI-Battery-Driver #21: Battery status not appearing on High Sierra 10.13

RehabMan, 2017-08-28:
On 10.13, battery status will not be detected at startup.

Workaround: At login screen, unplug AC adapter, plug back in (or opposite if starting on battery power).

It appears to be a timing issue.

Note: Sometimes it works, but it is rare.

RehabMan, 2017-08-30:
Current default for FirstPollDelay is 1000 (v1.81).
I think I may increase it later (perhaps to 3000 or 4000).
You can customize it easily with ACPI.

Andrey1970AppleLife, 2017-08-30:
v1.81 on my Acer, works with FirstPollDelay is 4000.

RehabMan, 2017-08-30:
I will probably change it to 4000 in v1.82.  And I may add the ability to change it with a kernel flag.
We'll see.

Strangely, at first 1000 was not working on my 4530s.  Then I increased it to 2000 (via ACPI RMFC override, so no kext cache rebuild), and it still didn't work. Then increased it to 3000, and it worked.  So then I reduced to 1000 again, and it still worked!  Reduced to 500... worked.  Reduced to 250, stopped working.  Went back to 1000 and it worked.

It was strange that I needed to "boot strap" it with a relatively large delay, but then after it was working, I could go lower, even to the breaking point, at which I didn't need the large value anymore (as 1000 worked after breaking it with 250).

Very strange timing issue for sure...

RehabMan, 2017-08-30:
version 1.8.2 seems to be working here... (both in 10.12 and 10.13).
More testing required though, especially with older versions.

RehabMan, 2017-08-30:
Tested 1.81.2 and working:
- HP ProBook 4540s, 10.13, 10.12
- HP ProBook 4530s, 10.13, 10.11, 10.10, 10.6.3.
- Lenovo u430, 10.12, 10.13
- Lenovo 710-11ISK, 10.13

RehabMan, 2017-08-30:
Fixed in 1.81.2.

closed 2017-08-30 by RehabMan

edward-p, 2017-09-28:
v1.81.3 on my MSI GP62 (installed on HDD), I need to increase FirstPollDelay to 6000 on 10.13. That's a little bit high, isn't it?

RehabMan, 2017-09-28:
Probably due to slow boot caused by APFS + TRIM.
If you're using APFS, disable the TRIM patch.

edward-p, 2017-09-28:
@RehabMan 
Solved by disable the TRIM patch. Thanks for reply.

edward-p, 2017-09-28:
Actually I Installed High Sierra on my HDD not SSD. I'm using APFS on HDD and applied TRIM patch to my other SSDs. I tried disabling the TRIM patch. It works sometimes, but rare. So I guess there may be another issue caused the slow boot after I update to 10.13.

edward-p, 2017-10-05:
Seems this issue occurs again 😣, when I boot with AC Adapter Plug in.

I got many WARNINGs like this:

``` bash
$ log show | grep ACPIBatter
2017-10-05 02:11:29.775043+0800 0x1991     Default     0x0                  0      0    kernel: (kernel) ACPIBatteryManager: WARNING! fMaxCapacity > fDesignCapacity. adjusted fMaxCapacity from 3941, to 3924
2017-10-05 02:11:59.884658+0800 0x1b54     Default     0x0                  0      0    kernel: (kernel) ACPIBatteryManager: WARNING! fMaxCapacity > fDesignCapacity. adjusted fMaxCapacity from 3941, to 3924
2017-10-05 02:12:29.995633+0800 0x1db8     Default     0x0                  0      0    kernel: (kernel) ACPIBatteryManager: WARNING! fMaxCapacity > fDesignCapacity. adjusted fMaxCapacity from 3941, to 3924
2017-10-05 02:13:00.105222+0800 0x1db8     Default     0x0                  0      0    kernel: (kernel) ACPIBatteryManager: WARNING! fMaxCapacity > fDesignCapacity. adjusted fMaxCapacity from 3941, to 3924
2017-10-05 02:13:30.214448+0800 0x208f     Default     0x0                  0      0    kernel: (kernel) ACPIBatteryManager: WARNING! fMaxCapacity > fDesignCapacity. adjusted fMaxCapacity from 3941, to 3924
2017-10-05 02:14:00.321200+0800 0x22a2     Default     0x0                  0      0    kernel: (kernel) ACPIBatteryManager: WARNING! fMaxCapacity > fDesignCapacity. adjusted fMaxCapacity from 3941, to 3924
...
```
And battery status not appearing :(

kikiwora, 2018-02-01:
Well, on Hight Sierra 10.13.3 (APFS on SSD) if you've used trimforce enable (or some other TRIM patch?), battery status indicator is automatically turned off after boot (on logon screen and after login), you should enable it manually after every boot.

I assume, that it can be solved with abm_firstpolldelay boot flag, but for me the simplest solution was to turn off TRIM for a better times :)

P.S. for **edward-p**: Hight Sierra with APFS on HDD is a very bad idea. You should force install|update with HFS+J instead.

RehabMan, 2018-02-01:
> Well, on Hight Sierra 10.13.3 (APFS on SSD) if you've used trimforce enable (or some other TRIM patch?), battery status indicator is automatically turned off after boot (on logon screen and after login), you should enable it manually after every boot.

It is due to boot delays caused by problems with non-Apple SSDs and TRIM.
On NVMe drives, TRIM is always enabled, so you don't even have that option.
Maybe someone will figure out a patch, but for now, disabling TRIM or using HFS+J instead covers most of the problems.
And still, you can increase the delay with abm_firstpolldelay, which might solve the problem once you find the right value.  But likely you have other problems due to the boot delay anyway.

ouija, 2018-04-12:
Have similar issue occurring on an Asus S400CA (VivoBook) w/High Sierra where the battery icon won't appear in the status bar;  Tried setting polling both dsdt / boot arg methods for various timings up to 16 seconds, but still having an issue where, upon first booting, the battery icon will appear for a brief second in the status bar (if at all) and then won't appear unless go to Energy Saver in SysPrefs and it will (usually) say battery is 0% (even though fully charged) and if you click to enable it in the status bar, it won't stay clicked, UNLESS you physically unplug the power cord and plug it back in.

Then you can click to enable it in the status bar, and it will stay there until you next reboot (and do it all over again).

Any thoughts on what might fix this?  I tried delaying the startup by a couple seconds via the DSDT patch, but it didn't seem to have any effect.

dedicahyadi, 2018-04-13:
Sorry sir, i'm facing same problem like Ouija above, when battery full 100% and restarting with plug in adaptor, the battery not showing on status bar,,
Asus A456URK core i5 Kabylake with High Sierra 10.13.4
thanks