RehabMan/OS-X-ACPI-Battery-Driver #22: Low battery warning with 1.81.2 Sierra

srscott, 2017-09-01:
Kext installed to S/L/E
DSDT appropriately patched
Kext working perfectly in High Sierra
Kext in Sierra displays the battery percentage correctly but gives a low battery warning.

[image]

[image]

[DSDT.aml.zip](https://github.com/RehabMan/OS-X-ACPI-Battery-Driver/files/1268932/DSDT.aml.zip)

RehabMan, 2017-09-01:
I have this happening intermittently as well.
It happens only when booting on battery, not when AC is plugged (important data to mention).

I will probably need to make more of the 10.13 changes conditional such that the old code path is taken prior to 10.13.

You can install an older version on 10.12 in the interim.

RehabMan, 2017-09-01:
Try ACPIBatteryManager v1.81.3

srscott, 2017-09-02:
ACPIBatteryManager v1.81.3 has resolved the issue.

RehabMan, 2017-09-02:
Seems fixed then.
Closing.

closed 2017-09-02 by RehabMan