RehabMan/OS-X-ACPI-Battery-Driver #26: Couldnt show battery when startup...

gujiangjiang, 2017-10-27:
With the new kext it couldnt show battert icon when startup and must showit by myself and when restart it still disappear.

Old kext dont have this problem..

gujiangjiang, 2017-10-27:
I have emu+rc and nvram works but it still dont show icon when startup...

Shinji3rd, 2017-10-31:
I have this exact same problem.

TrungNguyen1909, 2017-11-04:
It also fail to update battery charge in realtime.

mrvir, 2017-11-08:
I have the same problem.
macOS 10.13.1 using the latest version of this kext.

Shinji3rd, 2017-12-04:
I saw that if you wait a couple of seconds before entering the password at the login screen, the battery level suddenly appear and it works. But if you enter the password immediately after boot without giving time, the battery level does not appear and you have to enable it manually.

TrungNguyen1909, 2017-12-20:
Try this! Worked for me!!!
[ACPIBatteryManager.kext.zip](https://github.com/RehabMan/OS-X-ACPI-Battery-Driver/files/1576112/ACPIBatteryManager.kext.zip)

RehabMan, 2017-12-24:
Use the forums for troubleshooting.

closed 2017-12-24 by RehabMan