RehabMan/OS-X-ACPI-Battery-Driver #29: `_BIF` causes cursor jumps with VoodooI2C

scottsanett, 2018-05-04:
I think I've found a workaround for this. The conflict is in this code in AppleSmartBattery.cpp:

```cpp
bool AppleSmartBattery::pollBatteryState(int path)
{
    // ...
    fProvider->getBatterySTA();
    if (fBatteryPresent)
    {
        if (fStartupFastPoll || path == kNewBatteryPath) {
            if (fUseBatteryExtendedInformation)
                fProvider->getBatteryBIX();
            else
                fProvider->getBatteryBIF();
            if (fUseBatteryExtraInformation)
                fProvider->getBatteryBBIX();
    }
    fProvider->getBatteryBST();
    // ...
}
```

What happens is whenever the poll starts, `_BIF` is called and causes a brisk interruption in the movement of the cursor, and if the cursor is being moved when this happens, a leap/jump occurs.

I wonder if `_BIF` is really necessary for a frequent poll like this, so my proposal is to add a statement for 
these taxing calls only when ```fStartupFastPoll || path == kNewBatteryPath``` . 

I'm testing it right now and have yet observed any side effects.

RehabMan, 2018-08-21:
Probably a bug in the I2C kexts...
You using polling or interrupt in I2C?
New I2C kexts or old?

shwiseman, 2018-10-02:
@scottsanett I'm facing the same issue. Would you mind sharing an implementation of your patch while this is worked out? Thanks!

djlucas123456, 2019-11-02:
Hello.  goes this problem to solve as it is written in the first post?  or can you send me the edited kext?