RehabMan/OS-X-ACPI-Battery-Driver #35: More than two battery devices

soffyo, 2020-03-26:
What about machines that have more than just two battery devices? For example my asus zenbook ux331un has BAT0, BAT1 and BAT2. BAT0 located within both EC0 and H_EC while BAT1 and BAT2 are located under H_EC only. How should I apply this fix in my situation? I want to try this because I am having waking from sleep issues only when on battery power, with sleep working fine only when on ac power.

athlonreg, 2020-03-27:
Hello, can you provide your origin ACPI? And I want to know how many batteries will to show in windows.

soffyo, 2020-03-27:
on windows only BAT0 is showing.
[image]

Looking into ACPI, BAT1 and BAT2 are present only under H_EC while BAT0 is found within both EC0 and H_EC. You will find my origin folder with original aml's and already decompiled dsl's.
[origin.zip](https://github.com/RehabMan/OS-X-ACPI-Battery-Driver/files/4391792/origin.zip)

athlonreg, 2020-03-27:
> on windows only BAT0 is showing.
> [image]
>
> Looking into ACPI, BAT1 and BAT2 are present only under H_EC while BAT0 is found within both EC0 and H_EC. You will find my origin folder with original [...]

I checked your DSDT, the BAT0 BAT1 and BAT2 in H_EC is disabled so they are useless. EC0 is the part that really works. So you only need to patch BAT0 in EC0 to make battery work.

[image]

soffyo, 2020-03-27:
Okay, so I don't need to make ssdt-batc. My battery works already, I'm just trying to fix a kernel panic when waking from sleep only when on battery.

athlonreg, 2020-03-27:
> Okay, so I don't need to make ssdt-batc. My battery works already, I'm just trying to fix a kernel panic that occurs when waking from sleep only when on battery.

Can you try this SSDT for battery? I patched two method _BIX and BIFA. the ACPI Patch you need to use is in this SSDT.
[SSDT-OCBAT0-ASUS-ZenBook-UX331UN.dsl.zip](https://github.com/RehabMan/OS-X-ACPI-Battery-Driver/files/4392000/SSDT-OCBAT0-ASUS-ZenBook-UX331UN.dsl.zip)

soffyo, 2020-03-27:
I already use this patch. here is my SSDT-BAT.aml
[SSDT-BATT.zip](https://github.com/RehabMan/OS-X-ACPI-Battery-Driver/files/4394318/SSDT-BATT.zip)