RehabMan/OS-X-ACPI-Battery-Driver #36: Anyway to support UART batteries?

wdhwg001, 2020-05-17:
Hi,

Surface Gen 5 devices (Surface Book 2 / Surface Pro 6 / Surface Laptop, etc.) uses UART to fetch its battery status, which makes their DSDT does not contain "EmbeddedControl" at all.

However, in jakeday/linux-surface#28 , some legendary hackers finished the research and provided a proof of concept: https://gist.github.com/qzed/01a93568efb863f1b7887f0f375c03fc 

It then becomes a linux kernel patch ( jakeday/linux-surface#321 ). But before that, people are using cron jobs to run the script and update their battery status.

Is there a way we could achieve something similar? Or should we work with https://github.com/VoodooI2C/VoodooI2C as they may have some experiences dealing with UART?

Thanks!

davidkathy, 2021-03-25:
I have the same issue, the system can't detec the power supply input status.

aben-ab, 2022-01-24:
`
panic(cpu 1 caller 0xffffff80187d5bf3): Kernel trap at 0xffffff8018501082, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0xffffff99daaee000, CR3: 0x000000002e82a000, CR4: 0x00000000003626e0
RAX: 0x0000000000055500, RBX: 0xffffff99dab43000, RCX: 0x0000000000000300, RDX: 0x0000000000000800
RSP: 0xffffffd04fefb7d8, RBP: 0xffffffd04fefb7f0, RSI: 0xffffff99daaee000, RDI: 0xffffff99dab43500
R8:  0x0000000000000004, R9:  0x0000000000000001, R10: 0x00000000ffffffff, R11: 0x00000000e0000000
R12: 0xffffff99dab43000, R13: 0x0000000000000800, R14: 0xffffff99daaedb00, R15: 0x0000000000000000
RFL: 0x0000000000010206, RIP: 0xffffff8018501082, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0xffffff99daaee000, Error code: 0x0000000000000000, Fault CPU: 0x1, PL: 0, VF: 10

Panicked task 0xffffff99db061670: 119 threads: pid 0: kernel_task
Backtrace (CPU 1), panicked thread: 0xffffff90419e8000, Frame : Return Address
0xffffffd04fefb200 : 0xffffff8018685ffd 
0xffffffd04fefb250 : 0xffffff80187e6035 
0xffffffd04fefb290 : 0xffffff80187d5803 
0xffffffd04fefb2e0 : 0xffffff8018625a60 
0xffffffd04fefb300 : 0xffffff80186863cd 
0xffffffd04fefb420 : 0xffffff8018685b86 
0xffffffd04fefb480 : 0xffffff8018f16409 
0xffffffd04fefb4f0 : 0xffffff80187d5bf3 
0xffffffd04fefb670 : 0xffffff80187d58d8 
0xffffffd04fefb6c0 : 0xffffff8018625a60 
0xffffffd04fefb6e0 : 0xffffff8018501082 
0xffffffd04fefb7f0 : 0xffffff801a85e11e 
0xffffffd04fefb840 : 0xffffff801c950952 
0xffffffd04fefb880 : 0xffffff801a85d0f9 
0xffffffd04fefb970 : 0xffffff801a83e1d3 
0xffffffd04fefba00 : 0xffffff801a83d78c 
0xffffffd04fefba60 : 0xffffff801a66828c 
0xffffffd04fefbb20 : 0xffffff801a57c3b2 
0xffffffd04fefbbf0 : 0xffffff801a57a938 
0xffffffd04fefbc60 : 0xffffff801a57a4e1 
0xffffffd04fefbc90 : 0xffffff8018e5a0a0 
0xffffffd04fefbcf0 : 0xffffff801a57b620 
0xffffffd04fefbd60 : 0xffffff801c94ecb9 
0xffffffd04fefbdd0 : 0xffffff8018e2ebaa 
0xffffffd04fefbe30 : 0xffffff8018e2e715 
0xffffffd04fefbef0 : 0xffffff8018e2d710 
0xffffffd04fefbf50 : 0xffffff8018e30779 
0xffffffd04fefbfa0 : 0xffffff801862518e 
      Kernel Extensions in backtrace:
         as.lvs1974.AirportBrcmFixup(2.1.4)[D72C0F46-3F9C-3339-93E5-747ECF8B6C28]@0xffffff801c94b000->0xffffff801c95efff
            dependency: as.vit9696.Lilu(1.5.9)[B051298C-0FFC-32A7-B0D6-CB0DE0B793C0]@0xffffff801c5a9000->0xffffff801c5d3fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[897B72E0-B98F-30BA-8CB2-4E5E469CE4B6]@0xffffff801b1e7000->0xffffff801b211fff
         com.apple.driver.AirPort.BrcmNIC(1400.1.1)[DF790A78-11AE-336D-91B9-598FA83DB34B]@0xffffff801a54f000->0xffffff801aaa9fff
            dependency: com.apple.driver.corecapture(1.0.4)[F86FE48A-AF97-3FA9-965C-BC40CE3E28CA]@0xffffff801ba25000->0xffffff801ba3efff
            dependency: com.apple.driver.mDNSOffloadUserClient(1.0.1b8)[9B70EE1B-9CAB-3C79-BFDA-B70DC66A8450]@0xffffff801b1e0000->0xffffff801b1e4fff
            dependency: com.apple.iokit.IO80211FamilyLegacy(1200.12.2b1)[210E54A5-E336-3882-B54C-16D88FD128FB]@0xffffff801a3ee000->0xffffff801a534fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[58A9BF44-6CDC-3A24-A588-5D72E2378E2A]@0xffffff801af4d000->0xffffff801af63fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[897B72E0-B98F-30BA-8CB2-4E5E469CE4B6]@0xffffff801b1e7000->0xffffff801b211fff
            dependency: com.apple.iokit.IOSkywalkFamily(1.0)[F2724203-8AF4-39E2-841E-92B08446E426]@0xffffff801b2c6000->0xffffff801b30dfff

Process name corresponding to current thread (0xffffff90419e8000): kernel_task
Boot args: igfxrpsc=1 -igfxbls alcid=21 brcmfx-aspm=67 chunklist-security-epoch=0 -chunklist-no-rev2-dev chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
21D48

Kernel version:
Darwin Kernel Version 21.3.0: Wed Jan  5 21:37:58 PST 2022; root:xnu-8019.80.24~20/RELEASE_X86_64
Kernel UUID: 93729D02-FE6F-355B-BA76-BA930AA7103F
KernelCache slide: 0x0000000018400000
KernelCache base:  0xffffff8018600000
Kernel slide:      0x0000000018410000
Kernel text base:  0xffffff8018610000
__HIB  text base: 0xffffff8018500000
System model name: MacBookPro13,1 (Mac-473D31EABEB93F9B)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 4119670952
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x00000000f58d3865
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x00000007e7a89e0d 0x0000000000000000
Zone info:
Foreign   : 0xffffff802f028000 - 0xffffff802f035000
Native    : 0xffffff804183b000 - 0xffffffa04183b000
Readonly  : 0xffffff850e507000 - 0xffffff86a7e9b000
Metadata  : 0xfffffff01c3bc000 - 0xfffffff03c4e5000
Bitmaps   : 0xfffffff03c4e5000 - 0xfffffff040ce5000
last started kext at 2848721144: as.acidanthera.BrcmPatchRAM3	2.6.1 (addr 0xffffff801cc0e000, size 53248)
`

aben-ab, 2022-01-24:
`panic(cpu 1 caller 0xffffff80187d5bf3): Kernel trap at 0xffffff8018501082, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0xffffff99daaee000, CR3: 0x000000002e82a000, CR4: 0x00000000003626e0
RAX: 0x0000000000055500, RBX: 0xffffff99dab43000, RCX: 0x0000000000000300, RDX: 0x0000000000000800
RSP: 0xffffffd04fefb7d8, RBP: 0xffffffd04fefb7f0, RSI: 0xffffff99daaee000, RDI: 0xffffff99dab43500
R8:  0x0000000000000004, R9:  0x0000000000000001, R10: 0x00000000ffffffff, R11: 0x00000000e0000000
R12: 0xffffff99dab43000, R13: 0x0000000000000800, R14: 0xffffff99daaedb00, R15: 0x0000000000000000
RFL: 0x0000000000010206, RIP: 0xffffff8018501082, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0xffffff99daaee000, Error code: 0x0000000000000000, Fault CPU: 0x1, PL: 0, VF: 10

Panicked task 0xffffff99db061670: 119 threads: pid 0: kernel_task
Backtrace (CPU 1), panicked thread: 0xffffff90419e8000, Frame : Return Address
0xffffffd04fefb200 : 0xffffff8018685ffd 
0xffffffd04fefb250 : 0xffffff80187e6035 
0xffffffd04fefb290 : 0xffffff80187d5803 
0xffffffd04fefb2e0 : 0xffffff8018625a60 
0xffffffd04fefb300 : 0xffffff80186863cd 
0xffffffd04fefb420 : 0xffffff8018685b86 
0xffffffd04fefb480 : 0xffffff8018f16409 
0xffffffd04fefb4f0 : 0xffffff80187d5bf3 
0xffffffd04fefb670 : 0xffffff80187d58d8 
0xffffffd04fefb6c0 : 0xffffff8018625a60 
0xffffffd04fefb6e0 : 0xffffff8018501082 
0xffffffd04fefb7f0 : 0xffffff801a85e11e 
0xffffffd04fefb840 : 0xffffff801c950952 
0xffffffd04fefb880 : 0xffffff801a85d0f9 
0xffffffd04fefb970 : 0xffffff801a83e1d3 
0xffffffd04fefba00 : 0xffffff801a83d78c 
0xffffffd04fefba60 : 0xffffff801a66828c 
0xffffffd04fefbb20 : 0xffffff801a57c3b2 
0xffffffd04fefbbf0 : 0xffffff801a57a938 
0xffffffd04fefbc60 : 0xffffff801a57a4e1 
0xffffffd04fefbc90 : 0xffffff8018e5a0a0 
0xffffffd04fefbcf0 : 0xffffff801a57b620 
0xffffffd04fefbd60 : 0xffffff801c94ecb9 
0xffffffd04fefbdd0 : 0xffffff8018e2ebaa 
0xffffffd04fefbe30 : 0xffffff8018e2e715 
0xffffffd04fefbef0 : 0xffffff8018e2d710 
0xffffffd04fefbf50 : 0xffffff8018e30779 
0xffffffd04fefbfa0 : 0xffffff801862518e 
      Kernel Extensions in backtrace:
         as.lvs1974.AirportBrcmFixup(2.1.4)[D72C0F46-3F9C-3339-93E5-747ECF8B6C28]@0xffffff801c94b000->0xffffff801c95efff
            dependency: as.vit9696.Lilu(1.5.9)[B051298C-0FFC-32A7-B0D6-CB0DE0B793C0]@0xffffff801c5a9000->0xffffff801c5d3fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[897B72E0-B98F-30BA-8CB2-4E5E469CE4B6]@0xffffff801b1e7000->0xffffff801b211fff
         com.apple.driver.AirPort.BrcmNIC(1400.1.1)[DF790A78-11AE-336D-91B9-598FA83DB34B]@0xffffff801a54f000->0xffffff801aaa9fff
            dependency: com.apple.driver.corecapture(1.0.4)[F86FE48A-AF97-3FA9-965C-BC40CE3E28CA]@0xffffff801ba25000->0xffffff801ba3efff
            dependency: com.apple.driver.mDNSOffloadUserClient(1.0.1b8)[9B70EE1B-9CAB-3C79-BFDA-B70DC66A8450]@0xffffff801b1e0000->0xffffff801b1e4fff
            dependency: com.apple.iokit.IO80211FamilyLegacy(1200.12.2b1)[210E54A5-E336-3882-B54C-16D88FD128FB]@0xffffff801a3ee000->0xffffff801a534fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[58A9BF44-6CDC-3A24-A588-5D72E2378E2A]@0xffffff801af4d000->0xffffff801af63fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[897B72E0-B98F-30BA-8CB2-4E5E469CE4B6]@0xffffff801b1e7000->0xffffff801b211fff
            dependency: com.apple.iokit.IOSkywalkFamily(1.0)[F2724203-8AF4-39E2-841E-92B08446E426]@0xffffff801b2c6000->0xffffff801b30dfff

Process name corresponding to current thread (0xffffff90419e8000): kernel_task
Boot args: igfxrpsc=1 -igfxbls alcid=21 brcmfx-aspm=67 chunklist-security-epoch=0 -chunklist-no-rev2-dev chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
21D48

Kernel version:
Darwin Kernel Version 21.3.0: Wed Jan  5 21:37:58 PST 2022; root:xnu-8019.80.24~20/RELEASE_X86_64
Kernel UUID: 93729D02-FE6F-355B-BA76-BA930AA7103F
KernelCache slide: 0x0000000018400000
KernelCache base:  0xffffff8018600000
Kernel slide:      0x0000000018410000
Kernel text base:  0xffffff8018610000
__HIB  text base: 0xffffff8018500000
System model name: MacBookPro13,1 (Mac-473D31EABEB93F9B)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

[... 100 lines cut ...]
|IOSCSIBlockCommandsDevice	452.60.2
|IO!S!F	2.1
|IOSCSIArchitectureModel!F	452.60.2
>!AMobileFileIntegrity	1.0.5
$!AImage4	4.2.0
@kext.CoreTrust	1
>!AFDEKeyStore	28.30
>!AEffaceable!S	1.0
>!ACredentialManager	1.0
>KernelRelayHost	1
|IOUSBHost!F	1.2
>!UHostMergeProperties	1.2
>usb.!UCommon	1.0
>!ABusPower!C	1.0
>!ASEPManager	1.0.1
>IOSlaveProcessor	1
>!AACPIPlatform	6.1
>!ASMC	3.1.9
|IOPCI!F	2.9
|IOACPI!F	1.4
>watchdog	1
@kec.pthread	1
@kec.Libm	1
@kec.corecrypto	12.0
`